microchip-pic-avr-tools/pymcuprog #1: Unable to write to userrow when device is locked

mechatroniks-git, 2021-05-24:
The datasheet claims I can write to userrow when device is locked. I've been unable to.

`pymcuprog -U -t uart -u COM8 -d attiny1624 write -m user_row -l 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
`
Same issue with this method
```
pymcuprog -t uart -u COM8 -d attiny1624 write -f userrow.hex -U
pymcuprog.serialupdi.link - WARNING - Check failed
pymcuprog.serialupdi.application - ERROR - Timeout waiting for device to unlock
pymcuprog.nvm - ERROR - Device is locked.
Error:
Failed to enter NVM programming mode: device is locked
pymcuprog.serialupdi.application - ERROR - Timeout waiting for device to unlock
pymcuprog.nvm - ERROR - Device is locked.
Error:
Failed to enter NVM programming mode: device is locked
Pinging device...
pymcuprog.pymcuprog_main - ERROR - Error with st_ptr
Done.
```

contents of userrow.hex (which progams OK when not locked)

```
:02000004008575
:10000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00
:10001000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0
:00000001FF
```

xedbg, 2021-06-01:
Bug logged (internal) as DSG-2859.
(Sorry - I had some issues with notifications...)

xedbg, 2021-06-07:
Hi @mechatroniks-git 
Turns out this was not implemented at all on serialupdi.
Implementation is now added, and queued for the next release.
In the meantime you can try version 3.9.2.123 on pypi test server.
pip install -i https://test.pypi.org/simple/ pymcuprog
(Bench-tested with AVR DA only at this point.)

xedbg, 2021-11-01:
Now fixed in 3.10.2
https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/CHANGELOG.md

closed 2021-11-01 by xedbg