microchip-pic-avr-tools/pymcuprog #2: Unable to write fuses on Atmega 4809 using serial updi

JMare, 2021-06-25:
Hi,

Not sure if I have found a bug or I am just doing it wrong, but I cant work this out from the docs or any examples I have found online.

When I use the API to write fuses using:

```
xfuses = [0x00,0xE4,0x02,0xFF,0xFF,0xC5,0x07,0x02,0xFF,0x00]

backend.write_memory(xfuses,MemoryNames.FUSES,0x00)
```

I get the following error 

```
Traceback (most recent call last):
  File "loader.py", line 39, in <module>
    backend.write_memory(xfuses,MemoryNames.FUSES,0x00)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/backend.py", line 529, in write_memory
    self.programmer.write_memory(data=data, memory_name=memory_name, offset=offset_byte)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/programmer.py", line 187, in write_memory
    self.device_model.write(memory, offset, data)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/nvmserialupdi.py", line 171, in write
    self.avr.nvm.write_flash(offset_aligned, chunk)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/nvm.py", line 223, in write_flash
    return self.write_nvm(address, data, use_word_access=True)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/nvm.py", line 301, in write_nvm
    self.readwrite.write_data_words(address, data)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/readwrite.py", line 138, in write_data_words
    return self.datalink.st_ptr_inc16(data)
  File "/home/james/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/link.py", line 142, in st_ptr_inc16
    constants.UPDI_DATA_16, data[0], data[1]])
IndexError: list index out of range
```

When I use PDB to look at the frame with the out of range exception, I see that it is trying to access data[1] of a since byte of data.

I get the same problem when I try and use a bytearray instead of a list.
Can anyone suggest what I am doing wrong?

(Pinging, erasing, writing flash and verifying all work fine, so my connection to the device is good and the api is otherwise working well)

xedbg, 2021-06-25:
logged internally: DSG-3538

xedbg, 2021-06-28:
Ouch. 
Sloppy if-elif here: https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/nvmserialupdi.py#L163
Will be fixed in the next release.

xedbg, 2021-06-28:
Available for test:
pip install -i https://test.pypi.org/simple/ pymcuprog==3.9.3.124

erazor83, 2021-07-20:
FYI
I got a similar issue when writing .fuse from ihex:
```
oipiopioWriting fuses...
pymcuprog.programmer - INFO - Write...
pymcuprog.programmer - INFO - Writing 9 bytes of data to fuses...
pymcuprog.nvm - DEBUG - Writing 1 bytes to address 0x001280
pymcuprog.serialupdi.nvm - DEBUG - Wait flash ready
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x4, 0x2, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x0]
pymcuprog.serialupdi.nvm - DEBUG - Load NVM address
pymcuprog.serialupdi.link - INFO - ST to 0x001008
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x44, 0x8, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.physical - DEBUG - send : [0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.link - INFO - ST to 0x001009
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x44, 0x9, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.physical - DEBUG - send : [0x12]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.nvm - DEBUG - Load fuse data
pymcuprog.serialupdi.link - INFO - ST to 0x001006
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x44, 0x6, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.physical - DEBUG - send : [0x0]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.nvm - DEBUG - Execute fuse write
pymcuprog.serialupdi.nvm - DEBUG - NVMCMD 7 executing
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x44, 0x0, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.physical - DEBUG - send : [0x7]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.nvm - DEBUG - Wait flash ready
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x4, 0x2, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x0]
pymcuprog.serialupdi.nvm - DEBUG - Wait flash ready
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x4, 0x2, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x0]
pymcuprog.serialupdi.nvm - DEBUG - Clear page buffer
pymcuprog.serialupdi.nvm - DEBUG - NVMCMD 4 executing
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x44, 0x0, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.physical - DEBUG - send : [0x4]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.nvm - DEBUG - Wait flash ready
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x4, 0x2, 0x10]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x0]
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x69, 0x80, 0x12]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.link - DEBUG - Repeat 0
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xa0, 0xff]
pymcuprog.serialupdi.link - DEBUG - ST16 to *ptr++
pymcuprog.pymcuprog_main - ERROR - Operation failed with IndexError: array index out of range
pymcuprog.pymcuprog_main - DEBUG - array index out of range
Traceback (most recent call last):
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/pymcuprog_main.py", line 92, in pymcuprog
    _programming_actions(backend, args)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/pymcuprog_main.py", line 447, in _programming_actions
    status = _action_write(backend, args)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/pymcuprog_main.py", line 339, in _action_write
    _write_memory_segments(backend, result, args.verify)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/pymcuprog_main.py", line 380, in _write_memory_segments
    backend.write_memory(segment.data, memory_name, segment.offset)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/backend.py", line 529, in write_memory
    self.programmer.write_memory(data=data, memory_name=memory_name, offset=offset_byte)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/programmer.py", line 187, in write_memory
    self.device_model.write(memory, offset, data)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/nvmserialupdi.py", line 171, in write
    self.avr.nvm.write_flash(offset_aligned, chunk)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/nvm.py", line 223, in write_flash
    return self.write_nvm(address, data, use_word_access=True)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/nvm.py", line 301, in write_nvm
    self.readwrite.write_data_words(address, data)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/readwrite.py", line 138, in write_data_words
    return self.datalink.st_ptr_inc16(data)
  File "/home/erazor/.local/lib/python3.8/site-packages/pymcuprog/serialupdi/link.py", line 142, in st_ptr_inc16
    constants.UPDI_DATA_16, data[0], data[1]])
IndexError: array index out of range
pymcuprog.serialupdi.application - INFO - Leaving NVM programming mode
pymcuprog.serialupdi.application - INFO - Apply reset
```

pymcuprog==3.9.3.124 works for me.

farfarfromhome, 2021-09-05:
I had a similar issue with an attiny614:
`pymcuprog -t uart -u COM9 -c 9600 -d attiny1614 write -m fuses -o 2 -l 1
pymcuprog.serialupdi.link - WARNING - Check failed
Pinging device...
Ping response: 1E9422
Writing literal values...
pymcuprog.pymcuprog_main - ERROR - Operation failed with IndexError: bytearray index out of range
Done.
`
Interim release 3.9.3.124 fixed it!

xedbg, 2021-11-01:
Fixed in 3.10.2
https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/CHANGELOG.md

closed 2021-11-01 by xedbg