microchip-pic-avr-tools/pymcuprog #4: attiny424 is not supported

mraardvark, 2021-07-30:


xedbg, 2021-07-30:
Logged internally: DSG-3635

xedbg, 2021-08-02:
Added missing devices. Beta release on test.pypi:
pip install -i https://test.pypi.org/simple/ pymcuprog==3.9.4.126

SpenceKonde, 2021-09-01:
These are still missing from the github repo... I wound up synthesizing them from other device definitions for megaTinyCore 2.4.0.

xedbg, 2021-09-01:
Sorry about that - I will look at adjusting the QA process to make releasing to github more frequent - right now it it only downstream from the pypi production release (not test.pypi) which has some added testing requirements.

SpenceKonde, 2021-09-01:
Aaaha, yeah - I would think the natural thing to do would be for the github repo to be updated with the latest code; Github has a "release" feature that is designed for declaring a snapshot of the repo to be production grade.

xedbg, 2021-11-01:
tiny424 and friends are now in 3.10.2
https://github.com/microchip-pic-avr-tools/pymcuprog/tree/main/pymcuprog/deviceinfo/devices

closed 2021-11-01 by xedbg