microchip-pic-avr-tools/pymcuprog #5: PyedbglibError: Unable to SET (failure code 0x11)

kristianlm, 2021-08-08:
Hi,

I'm trying to connect my Atmel-ICE to my new AVR128DB28 DIP chips on a breadboard. I've never used `pymcuprog` before, and I'm getting an error message that I don't know how to debug:

```
klm@pisa ~> sudo pip show pymcuprog | head -n2
Name: pymcuprog
Version: 3.9.1.120
ERROR: Pipe to stdout was broken
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
BrokenPipeError: [Errno 32] Broken pipe
klm@pisa ~> sudo pip show pyedbglib | head -n2
Name: pyedbglib
Version: 2.18.2.75
ERROR: Pipe to stdout was broken
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
BrokenPipeError: [Errno 32] Broken pipe
klm@pisa ~> pacman -Qi libusb|head -n2
Name            : libusb
Version         : 1.0.24-2
klm@pisa ~ [1]> sudo pymcuprog ping -d avr128db28
Connecting to anything possible
Connected to Atmel-ICE CMSIS-DAP from ATMEL (serial number J427000*****)
Debugger firmware version 1.0.33
Debugger hardware revision 0
Traceback (most recent call last):
  File "/usr/bin/pymcuprog", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/pymcuprog/pymcuprog.py", line 249, in main
    return pymcuprog_main.pymcuprog(arguments)
  File "/usr/lib/python3.9/site-packages/pymcuprog/pymcuprog_main.py", line 80, in pymcuprog
    status = _start_session(backend, device_selected, args)
  File "/usr/lib/python3.9/site-packages/pymcuprog/pymcuprog_main.py", line 548, in _start_session
    backend.start_session(sessionconfig)
  File "/usr/lib/python3.9/site-packages/pymcuprog/backend.py", line 363, in start_session
    self.programmer.setup_device(
  File "/usr/lib/python3.9/site-packages/pymcuprog/programmer.py", line 78, in setup_device
    self.device_model = get_nvm_access_provider(self.transport,
  File "/usr/lib/python3.9/site-packages/pymcuprog/nvm.py", line 45, in get_nvm_access_provider
    accessprovider = NvmAccessProviderCmsisDapUpdi(transport, device_info=device_info,
  File "/usr/lib/python3.9/site-packages/pymcuprog/nvmupdi.py", line 55, in __init__
    self.avr.setup_prog_session(khz=frequency // 1000, use_hv=use_hv)
  File "/usr/lib/python3.9/site-packages/pymcuprog/avr8target.py", line 190, in setup_prog_session
    self.protocol.set_variant(Avr8Protocol.AVR8_VARIANT_TINYX)
  File "/usr/lib/python3.9/site-packages/pyedbglib/protocols/avr8protocol.py", line 246, in set_variant
    self.set_byte(self.AVR8_CTXT_CONFIG, self.AVR8_CONFIG_VARIANT, variant)
  File "/usr/lib/python3.9/site-packages/pyedbglib/protocols/jtagice3protocol.py", line 278, in set_byte
    self._set_protocol(context, offset, bytearray([value]))
  File "/usr/lib/python3.9/site-packages/pyedbglib/protocols/jtagice3protocol.py", line 316, in _set_protocol
    raise PyedbglibError(msg)
pyedbglib.pyedbglib_errors.PyedbglibError: Unable to SET (failure code 0x11)

```

I'm getting this error whether my target MCU has power or not, so I suspect it's a software problem on my host machine. Otherwise, I would have expected a "target doesn't have power" error. Is this a correct assumption? I don't want to debug my wiring before I know that's causing the problem here.

I'm using Arch Linux. I've ran everything as root to avoid usb-permission problems as shown above, but things are still not working.

Any help would be appreciated,
Thank you

xedbg, 2021-08-09:
The debugger FW looks ancient (1.0.33, if that is accurate) - you should definitely upgrade using Studio/MPLAB.
Also: 
- check the linux notes on https://pypi.org/project/pyedbglib/
- we have not qualified pyedbglib for py3.9 just yet, so it might be worth trying in a 3.8 venv
- its recommended to install in "user space" (pip install --user pymcuprog)

kristianlm, 2021-08-09:
Thanks for your feedback, xedbg. I'm able to get the target voltage:

```
klm@pisa ~> pymcuprog getvoltage
Connecting to anything possible
Connected to Atmel-ICE CMSIS-DAP from ATMEL (serial number J427000*****)
Debugger firmware version 1.0.33
Debugger hardware revision 0
Measured voltage: 3.34V
```

So maybe python 3.9.6 is ok. I tried to hook up UPDI on my oscilloscope, but nothing is happening. I think I will try to upgrade the Atmel-ICE FW first. It is a bit strange though, as I bought the device not long ago. I'll post back when I find anything.

xedbg, 2021-08-09:
Well, most tools ship with "old" firmware, and the IDEs update on first use.
These devices are quite new, and thus need new firmware.

kristianlm, 2021-08-09:
I've tried installing `MPLAB_X_IDE_v5.50`, but I'm unable to find any firmware updating options from the menus. Is it possible to update the firmward using `pymcuprog`?

kristianlm, 2021-08-09:
Success! Now at 1.41.137 things seem to be working. Coming from atmega328p, I'm very excited to go from 6 to 3 pins for programming on my breadboards.

```
klm@pisa~ > pymcuprog -t atmelice -i updi -d avr128db28 ping
Connecting to any atmelice
Connected to Atmel-ICE CMSIS-DAP from ATMEL (serial number J427000*****)
Debugger firmware version 1.41.137
Debugger hardware revision 0
Pinging device...
Ping response: 1E970E
Done.
```

Thank you @xedbg for helping out!

xedbg, 2021-08-09:
Good to hear :)

closed 2021-08-10 by kristianlm