microchip-pic-avr-tools/pymcuprog #9: Attiny412 (and others) device info issues

TJALambert, 2021-11-04:
I'm currently unable to read the serial number for this device. With some digging it seems to be that the signatures section is set to a size of 0x3 currently, only allowing you access to the Device ID.

Changing `'signatures_size_bytes': 0x3` to `'signatures_size_bytes': 0x26` allows me to access the whole of the signatures as from the datasheet it should be from 0x00 to 0x25 (with 0x0D-0x1F reserved).
I believe this same issue is within multiple device info's as I had a quick look at some of the other attiny's and found the same 0x3 value set

xedbg, 2021-11-04:
Good catch @TJALambert - this is a similar symptom to the fuse size issue but somehow the opposite!
We harvest the size of the classic "signatures" (device ID) rather than the entire memory segment.  We will have to update the harvester to find the useful size including that serial number.
Thanks for the feedback!

xedbg, 2021-11-05:
Logged: DSG-3993

xedbg, 2022-05-24:
fixed in 3.13

closed 2022-05-24 by xedbg