microchip-pic-avr-tools/pymcuprog #12: Debug output from SerialUPDI uses non-padded hex strings

dbuchwald, 2021-11-26:
Sample output:

```
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc3, 0x8]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x02
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc2, 0x80]
```

Now, the problem is the `0x8` value, which should have been in fact 0x08. Other entries are not consistent with rest of the output either (`0xc3` here but `pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B` elsewhere). This is all due to usage of `hex(x)` function usage in `physical.py` module.

xedbg, 2021-11-26:
Thanks @dbuchwald !
Logged as DSG-4242
(We will take this in 'upstream' and it will pop out again on the next build published)

dbuchwald, 2021-11-26:
Thanks a lot!

closed 2021-11-26 by dbuchwald