microchip-pic-avr-tools/pymcuprog #19: pymcuprog.pymcuprog_main - ERROR - Operation failed with TypeError: a bytes-like object is required, not 'list'

kazkansouh, 2022-02-08:
I have observed the following:

```plaintext
$ pymcuprog -i updi -d atmega4809 --tool uart --uart /dev/ttyUSB0 read -m fuses -f fuses
Pinging device...
Ping response: 1E9651
Reading...
pymcuprog.pymcuprog_main - ERROR - Operation failed with TypeError: a bytes-like object is required, not 'list'
Done.
```

I believe the following line is at fault:

https://github.com/microchip-pic-avr-tools/pymcuprog/blob/a9411a8e4a5db8b54517c51da0bae96bf8385a65/pymcuprog/pymcuprog_main.py#L224

It should include a cast and be written as:

```python
data = bytes(item.data)
```

xedbg, 2022-02-08:
Thanks @kazkansouh
Logged internally as DSG-4488. Will take a look!

xedbg, 2022-05-24:
fixed in 3.13