microchip-pic-avr-tools/pymcuprog #20: Upload time

paulskirk53, 2022-02-08:
Hi Is there any way to make the upload faster? My 17kB code takes 33 seconds to load. My next project has significantly more code...
many thanks for providing this, I hope you can speed it up.

xedbg, 2022-02-08:
Have you tried the RC on test.pypi.org?

paulskirk53, 2022-02-08:
thanks, I don't know Python, so if I run

pip install pymcuprog, does that pick the latest version or do I have to download it?

this is what I get if I run pip install pymcuprog from a terminal in platformio:

```
Requirement already satisfied: pymcuprog in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (3.10.4.136)
Requirement already satisfied: pyyaml in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pymcuprog) (6.0)
Requirement already satisfied: pyedbglib>=2.19.5 in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pymcuprog) (2.19.6.89)
Requirement already satisfied: appdirs in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pymcuprog) (1.4.4)
Requirement already satisfied: pyserial in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pymcuprog) (3.4)
Requirement already satisfied: intelhex in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pymcuprog) (2.3.0)
Requirement already satisfied: hidapi in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pyedbglib>=2.19.5->pymcuprog) (0.10.1)
Requirement already satisfied: cython in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from pyedbglib>=2.19.5->pymcuprog) (0.29.22)
Requirement already satisfied: setuptools>=19.0 in c:\users\paul\appdata\local\programs\python\python39\lib\site-packages (from hidapi->pyedbglib>=2.19.5->pymcuprog) (49.2.1)
```

xedbg, 2022-02-08:
Ah, sorry.
pip install will fetch latest by default from "production pypi"
You can specify both an alternate index (like test pypi) and/or a specific version to fetch.  This will fetch latest from test.pypi:
pip install -i https://test.pypi.org/simple/ pymcuprog

paulskirk53, 2022-02-08:
thanks for your help, if i use that version, the upload fails:
this is the command line I use
pymcuprog write -d atmega4809 -t uart -u COM6 -c 250k -f .pio\build\ATmega4809\firmware.hex
with an AVR 4809 on com6

paulskirk53, 2022-02-08:
just get this:

```
PS C:\Users\Paul\Documents\PlatformIO\Projects\Live\Azimuth_Encoder> pymcuprog write -d atmega4809 -t uart -u COM6 -c 250k -f .pio\build\ATmega4809\firmware.hex
pymcuprog.serialupdi.link - WARNING - Check failed
pymcuprog.serialupdi.link - WARNING - Check failed
Traceback (most recent call last):
  File "c:\users\paul\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\paul\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python39\Scripts\pymcuprog.exe\__main__.py", line 7, in <module>
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\pymcuprog.py", line 285, in main
    return pymcuprog_main.pymcuprog(arguments)
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\pymcuprog_main.py", line 80, in pymcuprog
    status = _start_session(backend, device_selected, args)
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\pymcuprog_main.py", line 549, in _start_session
    backend.start_session(sessionconfig)
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\backend.py", line 363, in start_session
    self.programmer.setup_device(
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\programmer.py", line 78, in setup_device
    self.device_model = get_nvm_access_provider(self.transport,
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\nvm.py", line 44, in get_nvm_access_provider
    accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency, options=options)
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\nvmserialupdi.py", line 51, in __init__
    self.avr = UpdiApplication(port, baud, self.dut)
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\serialupdi\application.py", line 81, in __init__
    datalink.init_datalink()
  File "c:\users\paul\appdata\local\programs\python\python39\lib\site-packages\pymcuprog\serialupdi\link.py", line 60, in init_datalink
    raise PymcuprogError("UPDI initialisation failed")
pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed

```

xedbg, 2022-02-08:
Could you add in a -v info or -v debug to add logging verbosity and see if there are more breadcrumbs as to why this is failing?

paulskirk53, 2022-02-08:
ok, here is the output with -v info:

```
PS C:\Users\Paul\Documents\PlatformIO\Projects\Live\Azimuth_Encoder> pymcuprog write -d atmega4809 -t uart -u COM6 -c 250k -f .pio\build\ATmega4809\firmware.hex -v info
pymcuprog.programmer - INFO - Setting up programming session for 'atmega4809'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device atmega4809
pymcuprog.serialupdi.physical - INFO - Opening port 'COM6' at '250000' baud
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.application - INFO - SIB: 'megaAVR P:0D:1-3M2 (01.59B20.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'megaAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:1'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B20.0)'
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.application - INFO - PDI revision = 0x03
pymcuprog.serialupdi.application - INFO - Entering NVM programming mode
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.application - INFO - Release reset
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.application - INFO - SIB: 'megaAVR P:0D:1-3M2 (01.59B20.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'megaAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:1'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B20.0)'
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.application - INFO - PDI revision = 0x03
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.application - INFO - Device ID from pyupdi = '1E9651' rev 'B'
pymcuprog.nvm - INFO - Device family: 'megaAVR'
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.nvm - INFO - Device ID: '1E9651'
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.nvm - INFO - Device revision: '0.1'
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.nvm - INFO - Device serial number: 'b'515137414620b9f01f0a''
Ping response: 1E9651
Writing from hex file...
Writing flash...
pymcuprog.programmer - INFO - Write...
pymcuprog.programmer - INFO - Writing 11663 bytes of data to flash...
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
[... 509 lines cut ...]
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.programmer - INFO - Write complete.
pymcuprog.serialupdi.application - INFO - Leaving NVM programming mode
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.application - INFO - Release reset
Done.
pymcuprog.serialupdi.physical - INFO - Closing port 'COM6'

```

xedbg, 2022-02-08:
Well, that one looks like it succeeds - I was thinking about the failure case...

paulskirk53, 2022-02-08:
I'll keep the -v in the command line and if I get a failure again, I'll post. 

With this test version of pymcuprog, the upload time for the 17kB takes about 33 seconds.

xedbg, 2022-02-08:
OK, do that.
Also: performance can have many factors, many of which are in the serial adapter driver.  The nEDBG's CDC is not optimised for this kind of transfer at all, but it still manages 16k of a mega4809 in around 22sec on my setup.

paulskirk53, 2022-07-03:
Just to ask if there's any news on improving the upload speed? I am using the circuit as per diagram for upload on a 4809 chip.
When I previously used pyupdi (with the same UPDI circuit and equipment), the upload time was about half of what I'm getting now, so I guess it is possible to obtain faster upload? Thanks very much for considering this.

xedbg, 2022-07-04:
Hi @paulskirk53 - the other issue (https://github.com/microchip-pic-avr-tools/pymcuprog/issues/10) you posted about upload time was resolved as fixed in the 3.13 release in May.  The main culprit there was the ACK turnaround time - so ACK is now disabled.

I was not sure if this was a duplicate report, and left it open as it mentions a crash which appears to be non-reproducable. 

Maybe we need to re-sync:
- are you using 3.13?
- do you see a performance improvement over 3.10 (which was latest in february)?
- how does 3.13 compare to pyupdi in your setup?

paulskirk53, 2022-07-04:
Thanks, I just did an install and now find that on 11.5k of code, pymcuprog takes 22 seconds and pyupdi takes 32 seconds. So there is an improvement. I am not an electronics or programming professional and don't claim to understand much of the detail of this stuff. Perhaps UPDI is inherently slower than methods such as AVRISPs which I've used on other AVRs which are very fast. Thanks for providing pymcuprog, I appreciate it.

xedbg, 2022-07-04:
I think there is still a fair amount of overhead with the lengthy turnarounds via python - if speed becomes a concern, its always possible to pick up any one of the Curiosity Nano kits, saw off the target-end, and use it as a programmer.  From pymcuprog you just remove the -t and -u switches and it should just work...

paulskirk53, 2022-07-05:
thanks very much for the info, I'll see how it goes and I might try a sawn off nano :)

ygoe, 2024-05-16:
Not sure if this helps, but are you using a USB serial adapter? I always wondered how flashing a device is so slow for the tiny amount of data and only recently found a note that said that [FTDI adapters are preconfigured badly](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md#serial-adapter-requirements). You could set the latency to a lower value and it really did speed up things for me by a factor of 10 or so. Or use another USB serial adapter type.