microchip-pic-avr-tools/pymcuprog #21: Executable pymcuprog

kristofmulier, 2022-03-02:
Hi @xedbg ,
I noticed you're the most active one here in this repo. So I just tagged you to catch your attention. Hopefully you agree that some sort of build script (to produce an executable) would be a great improvement of `pymcuprog`. Please have a look at my pull request:

https://github.com/microchip-pic-avr-tools/pymcuprog/pull/18

Kind regards,
Kristof

xedbg, 2022-03-02:
Hi @kristofmulier,
Yes, I saw this, and sorry for not responding.
Our current strategy for binaries uses pyinstaller - as used in the iotprovision package - but we have not taken this as far as pymcuprog (yet).
So your contribution of an "alternative" option is appreciated, but not trivial to take in without proper consideration...

kristofmulier, 2022-03-04:
Hi @xedbg ,
I understand you prefer the `pyinstaller` option because you already use that in your other projects. I don't have any particular preference here. As long as the build system produces executables for Windows, Linux (and maybe also MAC), it's great :-)

Do you have an idea when you'd implement the `pyinstaller` build script?

Kind regards,
Kristof

xedbg, 2022-10-27:
Hi @kristofmulier 
I know its probably not what you were expecting... but the iotprovision binary released today can be 'skinned' into an executable pymcuprog, if that is useful.  
https://github.com/microchip-pic-avr-tools/iotprovision-bin