microchip-pic-avr-tools/pymcuprog #23: Where is the source code?

SammysHP, 2022-04-21:
At the time of writing this the repository has just a single commit from 1. November 2021. Where is the rest? I mean, there are older releases on PyPI and there are newer testing versions. Where can I find the source for all previous releases?

xedbg, 2022-04-21:
Hi @SammysHP,

GitHub source is published as a snapshot of an unzipped wheel - this is not optimal, but it is the workflow we are using for now.
To get 'previous source' you can always download the wheels and unzip them.
Anything in particular you are looking for??

SammysHP, 2022-04-21:
Thank you, @xedbg. This workflow makes forking and contributions from the community much more difficult because you have no hierarchy of commits that you can track, but on the other hand the community is rather small.

Nothing in particular, I was just confused that I couldn't find previous revisions or the latest development in this repository. If it is your policy to develop in private and just publish snapshots, that's fine, just a little bit weird.

closed 2022-04-21 by SammysHP

xedbg, 2022-04-21:
@SammysHP I understand and do agree that its not conducive to contributions.  
We rely on pymcuprog for regression test purposes for Curiosity Nanos, and that it still our primary use-case, which is reflected in that workflow.  (And also makes us less open to architectural changes which might be proposed.) Maybe one day we will adjust this though.

I have in the past taken PRs in by generating a patch, applying it to our bitbucket repo internally, and then the tweak gets released outward again - this is also how we use test.pypi as a way to publish changes which we can't test properly ourselves.