microchip-pic-avr-tools/pymcuprog #25: NO CMIS-DAP devices found

mcsmark, 2022-05-04:
Hi i use :
pymcuprog  ping
I get this output:
Connecting to anything possible
pyedbglib.hidtransport.hidtransportbase - ERROR - No CMSIS-DAP devices found.
Unable to connect to USB device

But the MLAB-SNAP is connected. with libusb i see this info
bus/device  idVendor/idProduct
bus-0/\\.\libusb0-0001--0x04d8-0x9018     04D8/9018
- Manufacturer : Microchip Technology Incorporated
- Product      : MPLAB® Snap ICD
so the device is connected.
what can be the problem?
Under windows device manager i see the Microchip WinUSB Device is working properly.

mcsmark, 2022-05-05:
i conctacted hidapi and they say it is not a hid device so their lib will not work. so i wonder is mlab-snap ever tested?

xedbg, 2022-05-05:
Yes, Snap is tested, and it _is_ a HID device if its in AVR / EDBG / CMSIS-DAP mode.
To get it into that mode, use Microchip Studio or MPLABX to make the switch by programming an AVR device.

mcsmark, 2022-05-12:
thanks, yes that was the solution. 
- install mlabx
- create an avr8 project
- try to program it
- the firmware will be updated/changed and you get a virtual COM port
- i also had to remove R48 resistor from the SNAP programmer board. 
- it is some work and i wonder why it is not simpler, but it works :-)
-

closed 2022-05-12 by mcsmark