microchip-pic-avr-tools/pymcuprog #27: pymcuprog reports upload success but is not uploading code

paulskirk53, 2022-07-04:
Hi, Apologies for a long post but the background is important.
I use a pair of AVR 4809 (40 pin variants) in a project. I use FTDI boards to facilitate code uploads and serial comms between a windows PC running my code and the AVRs.  I use the Platformio IDE. When I started with the 4809s it was my first experience of UPDI and I found pyupdi, which in those days was not deprecated. It worked fine as installed in my platformio IDE. I subsequently started using pymcuprog when pyupdi became deprecated, but as it is much slower to upload code than pyupdi, I reverted to pyupdi.
So the problem is that (for a reason I am exploring with the Platformio folks), pyupdi periodically disappears from my IDE. When this happens, I use pymcuprog as follows:

pymcuprog write -d atmega4809 -t uart -u COM3 -c 250k -f .pio\build\ATmega4809\firmware.hex

and it reports success when uploading code.
However, I don't think the code uploads. A simple blinkled prog. used as a trial does not blink the LED. Naturally to start with I thought it was me, but today I managed to reinstall pyupdi and used that to upload the same blinkled code. It worked, so just out of curiosity, I used pymcuprog to upload the code and that seemed to work too as reported below:

PS C:\Users\Paul\Documents\PlatformIO\Projects\Arduino-code-blinkled> pymcuprog write -d atmega4809 -t uart -u COM3 -c 250k -f .pio\build\ATmega4809\firmware.hex
Pinging device...
Ping response: 1E9651
Writing from hex file...
Writing flash...
Done.

But it doesn't.

So I think all the above can be summarised as follows:
If I use pyupdi to upload code, that works.
If I use pymcuprog, it doesn't work.

It's probably a setting I have missed. Thanks for any thoughts.

paulskirk53, 2022-07-04:
Ah, brill, thanks for the explanation.

paulskirk53, 2022-09-22:
Hi, I have just come back to this using the Platformio CLI, and using this command
pymcuprog write -d atmega4809 -t uart -u COM5 -c 250k -f .pio\build\ATmega4809\firmware.hex --erase
I get the error:
pymcuprog: error: unrecognized arguments: --erase

any help much appreciated.

xedbg, 2022-09-22:
@paulskirk53 - please verify which version of pymcuprog you are running since this --erase switch was added not long ago.
`pymcuprog --version`

paulskirk53, 2022-09-22:
it is 3.13.3.166
I did 
pip install pymcuprog

earlier today. It did work with the --erase option after I posted in July. 
thanks for help.

mraardvark, 2022-09-22:
That version was released in May... Are you sure there is not an older version in a venv inside platformio?

paulskirk53, 2022-09-22:
sorry I wouldn't know how to check, is there a command I can use to check for a virtual env.? Perhaps I should post about that on the PIO forum?

mraardvark, 2022-09-22:
I guess easiest would be to replace the programming flash write call in your .ini file with pymcuprog --version and check the output from that...

paulskirk53, 2022-09-22:
i was never able to get pymcuprog working via the platformio.ini file, so I use it from the CLI. So  I removed --erase and used this:
pymcuprog write -d atmega4809 -t uart -u COM5 -c 250k -f .pio\build\ATmega4809\firmware.hex --version

and it reports pymcuprog  version 3.13.3.166

hope that helps.

paulskirk53, 2022-09-23:
works fine this morning, no probs uploading with --erase. Only difference is the machine was restarted this morning. Version number is the same. Thanks for help.