microchip-pic-avr-tools/pymcuprog #28: Feature request: ability to set serial timeout when using serial UPDI

jzplusplus, 2022-07-15:
Hi, I am using pymcuprog to program a chip as soon as it is connected but running in a loop. The default timeout passed to pyserial in the "physical.py" file is 1 second. This leads to each failed programming attempt spending several seconds waiting and makes the program very slow to begin programming once the chip is actually present. A parameter passed to ToolSerialConnection would solve this problem. (I have tested setting the timeout to 0.1 and have not encountered any issues)

xedbg, 2022-08-09:
Logged internally as DSG-5158

xedbg, 2022-09-22:
@jzplusplus - if you want to give it a try:
pip install -i https://test.pypi.org/simple/ pymcuprog==3.13.12.4

xedbg, 2022-10-13:
This feature was added in 3.14 release.  Default behaviour is unchanged (although could probably have been reduced for the benefit of all...)