microchip-pic-avr-tools/pymcuprog #30: No connection to ATmega4809

StefMijs, 2022-08-19:
I try want to program an atmega4809 on a arduino nano every with an raspberry pi 4B with it's tx/rx pins.(The usb cable is no option)
I wave used the the following command to find the device but there is no response.
`
$ sudo pymcuprog ping -d atmega4809 -t uart -u  /dev/serial1 
Connecting to SerialUPDI
`
When i quit the command with ^C after a long time of waiting for response i get the following message:
`
^CTraceback (most recent call last):
  File "/usr/local/bin/pymcuprog", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/pymcuprog.py", line 292, in main
    return pymcuprog_main.pymcuprog(arguments)
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/pymcuprog_main.py", line 82, in pymcuprog
    status = _start_session(backend, device_selected, args)
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/pymcuprog_main.py", line 562, in _start_session
    backend.start_session(sessionconfig)
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/backend.py", line 376, in start_session
    self.programmer.setup_device(
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/programmer.py", line 82, in setup_device
    self.device_model = get_nvm_access_provider(self.transport,
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/nvm.py", line 44, in get_nvm_access_provider
    accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency, options=options)
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/nvmserialupdi.py", line 51, in __init__
    self.avr = UpdiApplication(port, baud, self.dut)
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/serialupdi/application.py", line 72, in __init__
    self.phy = UpdiPhysical(serialport, baud)
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/serialupdi/physical.py", line 31, in __init__
    self.send([constants.UPDI_BREAK])
  File "/usr/local/lib/python3.9/dist-packages/pymcuprog/serialupdi/physical.py", line 100, in send
    self.ser.write(command)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 622, in write
    abort, ready, _ = select.select([self.pipe_abort_write_r], [self.fd], [], None)
KeyboardInterrupt

`
When i use serial0 i get this massage:
`
pymcuprog.pymcuprog - ERROR - Operation failed with error: (22, 'Invalid argument')
`
I have connected it like the the following image. 

With kind regards.
[image]

Stef

xedbg, 2022-08-22:
Its been a while since I used the native serial on rPi, but it has worked in past (admittedly its not under regression test).
Have you followed the advice on the lower section ("Notes for Linux systems") on pyedbglib? (https://github.com/microchip-pic-avr-tools/pyedbglib)

StefMijs, 2022-09-13:
Just an update,
I have not found the solution yet, because the lag of time for this project at my job.
I want to thank you for your response.
If i find a solution i wil at it to the issue.

CameronRP, 2023-02-06:
@StefMijs I had this same issue on a RPi3 but have things working now. 
The issue was that I was using the `mini UART` instead of the `PL011 UART` (for why this is an issue look at "Notes for the Raspberry Pi 3.. " https://www.abelectronics.co.uk/kb/article/1035/serial-port-setup-in-raspberry-pi-os)
To fix this I just added `dtoverlay=disable-bt`  in `/boot/config.txt` and rebooted.