microchip-pic-avr-tools/pymcuprog #31: Read, Verify broken for ATtiny402

jcharapata, 2022-08-24:
Heyo, 
I've been trying to work with the ATtiny402 using an atmel ice programmer through updi interface. The MCU has a 4kb flash memory. However, reading more than 256 Bytes at a time causes the read_memory output to be corrupted/incorrect. For example, reading 257 bytes will corrupt the first 1-2 lines of the output data. This, in turn, messes with the Verify function as well. I've verified that programming works properly with Atmel studio 7.

Best I can tell, the values for ATtiny402 are accurate, but you probably know more about those parameters than I do. 
Haven't tried this with any other UPDI chips, but I've tried multiple 402's and multiple Atmel ICE programmers. 

Let me know if you have any thoughts

xedbg, 2022-08-25:
Logged internally as DSG-5242

xedbg, 2022-08-25:
Would you mind sharing some additional logs on this?  use for example the -v debug switch.
(I don't think the flash content will be exposed, just the packet details)

jcharapata, 2022-08-25:
`PS C:\Users> pymcuprog -C -m flash -d ATTiny402 read -o 0 -b 256 -V
pymcuprog version 3.13.3.166
PS C:\Users> pymcuprog -C -m flash -d ATTiny402 read -o 0 -b 256 -v debug
Connecting to anything possible
Connected to Atmel-ICE CMSIS-DAP from ATMEL (serial number J42700016055)
Debugger firmware version 1.39.130
pymcuprog.programmer - INFO - Setting up programming session for 'attiny402'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny402
pymcuprog.nvm - INFO - UPDI baud rate: 900000bps
pymcuprog.nvm - INFO - UPDI-specific initialiser
pymcuprog.nvm - INFO - Activating chip-erase key entry mechanism
pymcuprog.nvm - INFO - Key chip-erase complete
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (02.59B14.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(02.59B14.0)'
pymcuprog.nvm - INFO - PDI revision = 0x02
pymcuprog.nvm - INFO - Device ID: '1E9227'
pymcuprog.nvm - DEBUG - Device revision: 0x02
pymcuprog.nvm - INFO - Device revision: '0.2'
Ping response: 1E9227
Reading...
pymcuprog.programmer - INFO - Reading 256 bytes from flash...
pymcuprog.avr8target - INFO - Reading from address 0x000000
Memory type: flash
---------------------------------------------------------
0x008000: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008010: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008020: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008030: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x008040: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008050: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008060: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008070: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x008080: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008090: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080A0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080B0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x0080C0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080D0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080E0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080F0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
---------------------------------------------------------

pymcuprog.nvm - INFO - UPDI-specific de-initialiser
Done.
PS C:\Users> pymcuprog -C -m flash -d ATTiny402 read -o 0 -b 257 -v debug
Connecting to anything possible
Connected to Atmel-ICE CMSIS-DAP from ATMEL (serial number J42700016055)
Debugger firmware version 1.39.130
Debugger hardware revision 0
pymcuprog.programmer - INFO - Setting up programming session for 'attiny402'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny402
pymcuprog.nvm - INFO - UPDI baud rate: 900000bps
pymcuprog.nvm - INFO - UPDI-specific initialiser
pymcuprog.nvm - INFO - Activating chip-erase key entry mechanism
pymcuprog.nvm - INFO - Key chip-erase complete
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (02.59B14.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(02.59B14.0)'
pymcuprog.nvm - INFO - PDI revision = 0x02
pymcuprog.nvm - INFO - Device ID: '1E9227'
pymcuprog.nvm - DEBUG - Device revision: 0x02
pymcuprog.nvm - INFO - Device revision: '0.2'
Ping response: 1E9227
Reading...
pymcuprog.programmer - INFO - Reading 258 bytes from flash...
pymcuprog.avr8target - INFO - Reading from address 0x000000
Memory type: flash
---------------------------------------------------------
0x008000: 02 00 27 00 41 56 52 20 50 3A 30 44 3A 30 2D 33
0x008010: 4D 32 20 28 30 32 2E 35 39 42 31 34 2E 30 29 00
0x008020: 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008030: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x008040: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008050: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008060: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008070: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x008080: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x008090: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080A0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080B0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x0080C0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080D0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080E0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0080F0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

0x008100: 00 xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx
---------------------------------------------------------

pymcuprog.nvm - INFO - UPDI-specific de-initialiser
Done.
PS C:\Users>`

jcharapata, 2022-08-25:
This log is done on a confirmed erased chip. you can see on the second call with reading 257 bytes, the first few lines turn into nonsense

xedbg, 2022-08-25:
Hi @jcharapata 
I tried to reproduce this, but failed at first. Then I rolled back the atmel-ice firmware to the version that matches your output and it failed in the same way!
So: please update your atmel-ice firmware!
To do this you will need Atmel Studio's atfw.exe utility - download the latest pack from https://packs.download.microchip.com/ and run atfw.exe with -a <path to firmware zip>
Hope this works!

jcharapata, 2022-08-29:
This resolved my issue! Had to update my IDE, Atmel Studio said the programmer was up to date! 
Thanks for the assist!

closed 2022-08-31 by jcharapata

fulminemizzega, 2022-09-01:
Hello, sorry for bothering you on a closed issue, but I'm seeing the same thing with an attiny817-xmini board (attiny817 has flash page size of 64 bytes). This uses mEDBG version 1.0d (shown in atmel studio -> Tools -> Device Programming). I think this is the latest version available, using atfw with the corresponding atpack fails, I've found instead medbg_fw.zip in atmel studio tree and I can confirm this is firmware 1.0d:
```
PS C:\Program Files (x86)\Atmel\Studio\7.0\atbackend> .\atfw.exe -a "C:\Program Files (x86)\Atmel\Studio\7.0\tools\mEDBG\medbg_fw.zip" -t medbg
Found medbg:ATML2658061800009405
Upgrading medbg:ATML2658061800009405
Upgrading MCU firmware: [==========]
Successful upgrade
PS C:\Program Files (x86)\Atmel\Studio\7.0\atbackend> .\atfw.exe -r -t medbg
Found medbg:ATML2658061800009405
Master MCU Version: 1.d
```
In a linux (fedora 36) VM with pymcuprog programming works fine, reading 254 bytes is fine, reading 255 bytes corrupts the first page.
```
$ pymcuprog read -b 254 -m flash -d attiny817 -v debug
Connecting to anything possible
Connected to mEDBG CMSIS-DAP from Atmel (serial number ATML2658061800009405)
Debugger firmware version 1.13.43
Debugger hardware revision 0
Device mounted: 'attiny817'
pymcuprog.programmer - INFO - Setting up programming session for 'attiny817'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny817
pymcuprog.nvm - INFO - UPDI baud rate: 900000bps
pymcuprog.nvm - INFO - UPDI-specific initialiser
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (01.59B16.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B16.0)'
pymcuprog.nvm - INFO - PDI revision = 0x02
pymcuprog.nvm - INFO - Device ID: '1E9320'
pymcuprog.nvm - DEBUG - Device revision: 0x01
pymcuprog.nvm - INFO - Device revision: '0.1'
Ping response: 1E9320
Reading...
pymcuprog.programmer - INFO - Reading 254 bytes from flash...
pymcuprog.avr8target - INFO - Reading from address 0x000000
Memory type: flash
---------------------------------------------------------
0x008000: 19 C0 20 C0 1F C0 1E C0 1D C0 1C C0 1B C0 1A C0
0x008010: 19 C0 18 C0 17 C0 16 C0 15 C0 14 C0 13 C0 12 C0
0x008020: 11 C0 10 C0 0F C0 0E C0 0D C0 0C C0 0B C0 0A C0
0x008030: 09 C0 08 C0 11 24 1F BE CF EF CD BF DF E3 DE BF

0x008040: 02 D0 A6 C0 DD CF 88 ED 84 BF 81 E0 80 93 61 00
0x008050: 84 E0 80 93 21 04 88 E0 80 93 22 04 86 EB 92 E0
0x008060: 80 93 08 08 90 93 09 08 82 EC 80 93 06 08 83 E0
0x008070: 80 93 07 08 E8 EA F1 E8 82 ED 91 E8 21 91 30 91

0x008080: 04 08 35 FF FC CF 20 93 02 08 8E 17 9F 07 B1 F7
0x008090: 28 EA 31 E8 70 E2 5D E9 43 E0 60 E4 80 91 01 08
0x0080A0: 87 FF FC CF 90 91 00 08 96 36 51 F0 92 37 09 F4
0x0080B0: 60 C0 80 91 04 08 85 FF FC CF 90 93 02 08 EE CF

0x0080C0: E0 E0 F4 E8 80 81 81 93 E0 32 84 E8 F8 07 D1 F7
0x0080D0: 80 91 04 08 85 FF FC CF 70 93 02 08 80 91 01 08
0x0080E0: 87 FF FC CF 80 91 00 08 81 93 CF 01 8F 73 99 27
0x0080F0: 89 2B C1 F0 E4 38 FE 07 89 F7 80 81 81 93 xx xx
---------------------------------------------------------

pymcuprog.nvm - INFO - UPDI-specific de-initialiser
Done.
```
Compared to this:
```
$ pymcuprog read -b 255 -m flash -d attiny817 -v debug
Connecting to anything possible
Connected to mEDBG CMSIS-DAP from Atmel (serial number ATML2658061800009405)
Debugger firmware version 1.13.43
Debugger hardware revision 0
Device mounted: 'attiny817'
pymcuprog.programmer - INFO - Setting up programming session for 'attiny817'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny817
pymcuprog.nvm - INFO - UPDI baud rate: 900000bps
pymcuprog.nvm - INFO - UPDI-specific initialiser
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (01.59B16.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B16.0)'
pymcuprog.nvm - INFO - PDI revision = 0x02
pymcuprog.nvm - INFO - Device ID: '1E9320'
pymcuprog.nvm - DEBUG - Device revision: 0x01
pymcuprog.nvm - INFO - Device revision: '0.1'
Ping response: 1E9320
Reading...
pymcuprog.programmer - INFO - Reading 256 bytes from flash...
pymcuprog.avr8target - INFO - Reading from address 0x000000
Memory type: flash
---------------------------------------------------------
0x008000: 55 25 00 00 00 00 00 01 00 00 30 44 3A 30 2D 33
0x008010: 4D 32 20 28 30 31 2E 35 39 42 31 36 2E 30 29 00
0x008020: 00 00 00 00 00 14 00 13 00 11 80 12 8A 12 20 93
0x008030: 00 00 00 00 11 24 1F BE CF EF CD BF DF E3 DE BF

0x008040: 02 D0 A6 C0 DD CF 88 ED 84 BF 81 E0 80 93 61 00
0x008050: 84 E0 80 93 21 04 88 E0 80 93 22 04 86 EB 92 E0
0x008060: 80 93 08 08 90 93 09 08 82 EC 80 93 06 08 83 E0
0x008070: 80 93 07 08 E8 EA F1 E8 82 ED 91 E8 21 91 30 91

0x008080: 04 08 35 FF FC CF 20 93 02 08 8E 17 9F 07 B1 F7
0x008090: 28 EA 31 E8 70 E2 5D E9 43 E0 60 E4 80 91 01 08
0x0080A0: 87 FF FC CF 90 91 00 08 96 36 51 F0 92 37 09 F4
0x0080B0: 60 C0 80 91 04 08 85 FF FC CF 90 93 02 08 EE CF

0x0080C0: E0 E0 F4 E8 80 81 81 93 E0 32 84 E8 F8 07 D1 F7
0x0080D0: 80 91 04 08 85 FF FC CF 70 93 02 08 80 91 01 08
0x0080E0: 87 FF FC CF 80 91 00 08 81 93 CF 01 8F 73 99 27
0x0080F0: 89 2B C1 F0 E4 38 FE 07 89 F7 80 81 81 93 00 xx
---------------------------------------------------------

pymcuprog.nvm - INFO - UPDI-specific de-initialiser
Done.
```

xedbg, 2022-09-02:
Indeed you are right - it appears that this fix was made in the Atmel-ICE stack but not rolled out to the mEDBG... and this platform is no  longer under maintenance.  I think the solution here will have to involve doing the chunking by 128 (or just flash page size of 64) on the Python stack side rather than try to re-release that firmware...