microchip-pic-avr-tools/pymcuprog #32: Program write not working for AVR128DA MCU

nabelekt, 2023-01-02:
Hello, all. I have a project using an AVR128DA28 which I have always previously programmed using the Arduino IDE with [DxCore](https://github.com/SpenceKonde/DxCore). Here are the settings that I have there:

[image]

Here is the code I am experimenting with:
```
#include <avr/io.h>
#include <util/delay.h>

#define LED_PIN_YELLOW 7

int main (void)
{
  // init();

  // Setup pins
  PORTD.DIR = 0;  // Initialize direction for all Port D pins to input
  PORTD.DIR = PORTD.DIR | (1 << LED_PIN_YELLOW);  // Set yellow LED pin to be output

  PORTD.OUT = 0;  // Start with all LEDs off

  while (1) {

    PORTD.OUT = PORTD.OUT | (1 << LED_PIN_YELLOW);  // Turn on yellow LED
    _delay_ms(300);

    PORTD.OUT = PORTD.OUT & ~(1 << LED_PIN_YELLOW);  // Turn off yellow LED
    _delay_ms(600);

  }

  return 0;
}
```

If I flash using the Arduino IDE, my LED blinks as expected.

If I flash the MCU using the IDE and then run `pymcuprog read -t uart -u COM4 -d avr128da28 -f blink_read_arduino`, I get these files: [blink_read_arduino.zip](https://github.com/microchip-pic-avr-tools/pymcuprog/files/10329856/blink_read_arduino.zip).

If I then run `pymcuprog write -t uart -u COM4 -d avr128da28 -f blink.hex`. I get:
```
Connecting to SerialUPDI
Pinging device...
Ping response: 1E970A
Writing from hex file...
Writing flash...
Done.
```
but my LED does not flash. And `pymcuprog verify -t uart -u COM4 -d avr128da28 -f blink.hex` gives me:
```
Connecting to SerialUPDI
Pinging device...
Ping response: 1E970A
Verifying...
pymcuprog.programmer - ERROR - Verify failed for flash memory:
pymcuprog.programmer - ERROR - Verify mismatch starting at location 0x000000: 0x0C vs 0x00 (is the memory section erased?)
Done.
```
Here is the `blink.hex` file that I am flashing: [blink.txt](https://github.com/microchip-pic-avr-tools/pymcuprog/files/10329873/blink.txt) (renamed because I can't upload a .hex file here).

If I then run `pymcuprog read -t uart -u COM4 -d avr128da28 -f blink_read_cmd_line`, I get these files: [blink_read_cmd_line_hex.zip](https://github.com/microchip-pic-avr-tools/pymcuprog/files/10329865/blink_read_cmd_line_hex.zip).

I differed the files that I've attached and made the following notes:
- eeprom - identical
- flash - The first 268 bytes are different, but the rest are the same
- fuses - identical
- internal sram - Command line file has '0088 0088' repeated over and over; looks like the Arduino IDE file has real data. This is probably because nothing is running with the command line flash?
- lockbits - identical
- user row - identical
- signatures - identical

So, any ideas what I need to do differently to replicate what the Arduino IDE/DxCore is doing? It looks like that is actually using a modified version of pymcuprog.

Any help would be greatly appreciated. Thank you!

nabelekt, 2023-01-02:
I have found that if I add `--erase` to the end of my `write` command, the program works as expected. Why is the `--erase` required? If that is expected/desired, this issue can be closed.

xedbg, 2023-01-02:
You are not the first, and you won't be the last :|
(It's not as simple as it might seem to have a universal erase across AVR, PIC and SAM... in hindsight the default behaviour might has been better inverted, but we favour consistency at this point.)
Glad you got it to work!

closed 2023-01-02 by pic-avr-github-admin

nabelekt, 2023-01-02:
I think the reason that I wasn't using `--erase` in the first place was that the help makes it seem like it wouldn't apply to me:
`  --erase               erase memory section before writing (from an Intel(R) hex file only)`, particularly the part in parentheses. I'm not sure what an "Intel hex file" is but was thinking it was something that would be meant to be flashed to an Intel device.