microchip-pic-avr-tools/pymcuprog #33: Linux UART left in unknown state after first use flashing UPDI

bradanlane, 2023-01-18:
**Command:** `pymcuprog write -t uart -d attiny1616 -u /dev/ttyUSB0 -c 115200 -f  firmware.hex --erase --verify

**Platform:** Linux (Ubuntu 22.04.1 LTS)
**UART:** CH340-E
**Target:** ATtiny1616
**Behavior:** flash completes first time successfully; subsequent attempts fail.
```
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/ttyUSB0' at 115200 baud (timeout 1.0s)
pymcuprog.serialupdi.physical - ERROR - Unable to open serial port '/dev/ttyUSB0'
```
unplugging / pluggin-in the USB-UART accomplish some sort of reset and it will work again, once.

**Note:**  The same test under Windows 10 x64 works every time and does not require the USB-UART to bet manually reset.

_I recognize this may not be the result of `pymcuprog` and am open to suggestions._

xedbg, 2023-01-18:
Logged internally as DSG-6071

xedbg, 2023-01-18:
I have tested this with Linux Ubuntu 22.04.01 LTS and a mega4809 on a Curiosity Nano using its own CDC and it works just fine every time.
Do you have an alternative to the CH340-E to try for a second-opinion?

bradanlane, 2023-01-18:
Thanks for testing.

I have a CR2102 USB-UART. I will configure a test with it.

bradanlane, 2023-01-20:
I tested with both the CH340 and CR2102 and both have the issue.
However, if I run the command `sudo usbreset ...` I can use the device again and from then on it works (until I reboot).

I realized I left one thing out of my original description and it may be the underlying cause. The environment is "Linux (Ubuntu 22.04.1 LTS) in a virtual machine". It may well be the virtualization layer of the USB that is the actual issue.

Given the environment may be the actually issue, I would conclude `pymcuprog` does not have an issue and this should be closed.

I will leave this for a couple of days for anyone to add comments.

xedbg, 2023-01-23:
Thanks for the update!
Indeed, virtual machines using real hardware is a challenge.
Do you mind sharing more details about the virtualisation setup you are attempting? (eg: WSL?)

bradanlane, 2023-01-24:
My environment is Lubuntu 22.04 running in a VMware Workstation 15 image. 

I feel the issue is partly the VM. There is likely a “cheap USB-UART” complicating the situation.

The second use, after starting up the VM is _nearly_ always a problem. Some combination of `usbreset` and/or dropping/connecting at the VM level sorts things and then it works fine for the rest of the day.

xedbg, 2023-01-24:
Thanks - we use VMs for many scenarios but not  (yet) relating to hardware tools.
(FWIW: those symptoms remind me somewhat of a problem i encountered a while back using a VM in which the data-toggle bit was getting out of sync between a client and host on reconnect - i this case it was a 50/50 situation which depended on the number of packets in the session; and only one end resetting the toggle on the second connect...  that said, this was _quite_ some time ago, so I would hope things like this have been addressed... :)

bradanlane, 2023-03-01:
I've been using pymcuprog in a VM to a USM UART CH340 for a few weeks. The issue occurs frequently but not consistently.  Using `sudo usbreset ...` works sometimes but hangs more often than not.

thus far I do not have a reproducible set of steps. The problem is there but hard to pin down and I cannot say with 100% certainty if it is the VM, pymcuprog, or the intersection of the two.