microchip-pic-avr-tools/pymcuprog #34: Commandline option to explicitly set the DTR-line

wimalopaan, 2023-03-24:
For UPDI it would be very helpfull to have an cmd-line option to explicitly set the DTR-line of the attached serial device that is used for the UPDI. The DTR-line the should be held low or high as given on the cmd-line for the whole UPDI programming sequence.

Surely, this is kind of abuse of the DTR-line.

But this could be very helpfull in case one has a µC-board with two UPDI-capable µC on board. With this both µC could be programmed using the same serial line and without manual switching on the board.

Thanks.

xedbg, 2023-03-27:
Logged as DSG-6211

xedbg, 2023-03-27:
@wimalopaan - no, DSG is a MIcrochip internal Jira tracker

wimalopaan, 2023-03-27:
Ok, I wasn't aware that this is an official MicroChip project.

ygoe, 2024-05-16:
Is this actively being worked on? Is there even a target milestone or date? I could use this very soon and was under the impression that it already exists. I'm currently browsing the code and trying hard to follow the convoluted paths of where things go to and come from. But maybe I'll just add this feature to the old pyupdi tool instead.

xedbg, 2024-07-17:
hi @ygoe and @wimalopaan and sorry for the delayed response.
The serialUPDI implementation in pymcuprog is provided for convenience as a reference.  During development and test we use the onboard CDC of Curiosity Nanos looped back to its own UPDI line - this is practical since pymcuprog is a vital part of testing and QA of the Curiosity Nano platform for Microchip.
Unfortunately we can't afford the time to integrate all feature requests, specially those which may be dependent on custom hardware (like CDC/UART adapters).