microchip-pic-avr-tools/pymcuprog #35: Feature request: make info logging more concise

hb020, 2023-03-25:
Using pip version from pymcuprog, and serialupdi, in attiny 2 series (3226).
With "-v warning", there are no logs on the progress of the programming.
With "-v info", there are way too many log lines that are strictly identical and tell nothing:

```
....
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.application - INFO - PDI revision = 0x03
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.application - INFO - Device ID from pyupdi = '1E9651' rev 'B'
pymcuprog.nvm - INFO - Device family: 'megaAVR'
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.nvm - INFO - Device ID: '1E9651'
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.nvm - INFO - Device revision: '0.1'
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.nvm - INFO - Device serial number: 'b'515137414620b9f01f0a''
Ping response: 1E9651
Writing from hex file...
Writing flash...
pymcuprog.programmer - INFO - Write...
pymcuprog.programmer - INFO - Writing 11663 bytes of data to flash...
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to 0x001000
pymcuprog.serialupdi.link - INFO - LD from 0x001002
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.link - INFO - ST to 0x001000
[... 518 lines cut ...]
avrdude: device signature = 0x1e9527 (probably t3226)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .pio/build/ATtiny3226/firmware.hex for flash
         with 29047 bytes in 1 section within [0, 0x7176]
         using 227 pages and 9 pad bytes
avrdude: writing 29047 bytes flash ...

Writing | ################################################## | 100% 35.64s

avrdude: 29047 bytes of flash written
avrdude: verifying flash memory against .pio/build/ATtiny3226/firmware.hex

Reading | ################################################## | 100% 13.83s

avrdude: 29047 bytes of flash verified

avrdude done.  Thank you.
```

Much better. 

**Feature request:**
can the info logs be adapted to show concise progress info?

xedbg, 2023-03-27:
Logged as DSG-6210.
Very good input @hb020 
We have clearly not put that much thought into the logging... and avrdude is a great reference point, thanks!

pic-avr-github-admin, 2023-11-07:
Addressed in 3.16 release.

closed 2023-11-07 by pic-avr-github-admin