microchip-pic-avr-tools/pymcuprog #44: Corrupted flash memory writes for Atmega328p with ISP interface

araqioui, 2024-04-16:
While trying to execute :
```
pymcuprog write  -f executable.hex --erase --verify -t atmelice -d atmega328p -i isp
```

The command always fails at flash memory verification step:
```
Connecting to any atmelice
Connected to Atmel-ICE CMSIS-DAP from ATMEL (serial number J42700040265)
Debugger firmware version 1.45.7
Debugger hardware revision 0
pymcuprog.nvm - WARNING - 
pymcuprog.nvm - WARNING - AVR-ISP/SPI stack is in Alpha state
pymcuprog.nvm - WARNING - Expect some features to be missing
pymcuprog.nvm - WARNING - 
Pinging device...
Ping response: 1E950F
Erasing device before writing from hex file...
Writing from hex file...
Writing flash...
Verifying flash...
pymcuprog.programmer - ERROR - Verify failed for flash memory:
pymcuprog.programmer - ERROR - Verify mismatch starting at location 0x000000: 0x0C vs 0xFF (is the memory section erased?)
Verification failed!
```
The `mismatch location address` is not always the same.

xedbg, 2024-04-16:
I think your fix works... interesting though, the tool protocol also has a delay argument which is not correctly used.  This fix in pyedbglib's avrispprotocol.py could arguably be more "correct" when seen from the tool firmware:

Add parameter to write_flash_page:
```
# def write_flash_page(self, byte_address, data):
def write_flash_page(self, byte_address, data, delay_ms=5):
```

Then send in the delay to the tool firmware and set bit 6 to make use of it:
```
# command.extend([0x81])  # Page mode
# command.extend([0])  # Not used
command.extend([0x91])  # Write page with timed delay
command.extend([delay_ms])
```

araqioui, 2024-04-17:
@xedbg Thanks for the review and the proposed fix.
I can't help but agree with you on your suggested fix... Tested and works just perfectly :100: 
I opened a [#PR](https://github.com/microchip-pic-avr-tools/pyedbglib/pull/6) to bring your suggested fix to life in the next release hopefully :smiley: