microchip-pic-avr-tools/pymcuprog #45: Make it callable from the source code

ygoe, 2024-05-16:
I'm not a Python expert by far. I know the basic syntax and can find my way around and make a few changes. And I can't get this script to run on the command line on Windows.

My Python version is 3.11.4. I have already installed the dependencies (by extracting the info I understood from setup.cfg to a requirements.txt file). But how am I supposed to run this? When I do a little guessing and run `python pymcuprog.py` from the "pymcuprog" directory, I get this:

```
Traceback (most recent call last):
  File "C:\Source\Elektronik\pymcuprog\pymcuprog\pymcuprog.py", line 25, in <module>
    from . import pymcuprog_main
ImportError: attempted relative import with no known parent package
```

I don't know what's going on here. pyupdi has worked just fine like that. Is there some additional start or entry script missing in the code?

ygoe, 2024-05-17:
This is required to call the pymcuprog script from any directory:

```
set PYTHONPATH=C:\path\to\pymcuprog
python -m pymcuprog.pymcuprog ping -d attiny1614 ...
```

xedbg, 2024-05-21:
Hi @ygoe 
When you install Python you have the option to "add scripts folder to path" which is the trick you missed.
Many tools (like pymcuprog) install with a small executable in the scripts folder which will make it usable as a CLI.
Where the scripts folder is located depends on your Python version and your system - I would suggest finding it and adding it to your path.

xedbg, 2024-05-21:
you can just install it in "editable mode" by changing to your git source folder and using:
`pip install -e .`