microchip-pic-avr-tools/pymcuprog #49: MPLAB Snap and UPDI for attiny814 possible?

simonschmidt, 2025-02-08:
Hi!

I'm trying to use MPLAB Snap to program an attin814, but I have no luck.
I have a 1k pull-up on PA0/UPDI.

```
$ pymcuprog --version
pymcuprog version 3.17.3.45
$ pymcuprog ping -t snap -d attiny814 -i updi --verbose debug 
Connecting to any snap
Connected to MPLAB Snap ICD CMSIS-DAP from Microchip Technology Incorporated (serial number BUR210381107)
Debugger firmware version 1.10.249
Debugger hardware revision 1
pymcuprog.programmer - INFO - Setting up programming session for 'attiny814'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny814
pymcuprog.nvm - DEBUG - UPDI baud rate: 900000bps
pymcuprog.nvm - DEBUG - UPDI-specific initialiser
pyedbglib.protocols.jtagice3protocol - ERROR - JTAGICE3 error response code 0x1C: 'AVR8_FAILURE_FRAMING_ERROR: Framing error' 
pymcuprog.pymcuprog - ERROR - Operation failed with Jtagice3ResponseError: AVR8_FAILURE_FRAMING_ERROR: Framing error
pymcuprog.pymcuprog - DEBUG - AVR8_FAILURE_FRAMING_ERROR: Framing error
Traceback (most recent call last):
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/pymcuprog.py", line 307, in main
    return pymcuprog_main.pymcuprog(arguments)
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/pymcuprog_main.py", line 89, in pymcuprog
    status = _start_session(backend, device_selected, args)
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/pymcuprog_main.py", line 586, in _start_session
    backend.start_session(sessionconfig)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/backend.py", line 382, in start_session
    self.programmer.start(user_interaction_callback=user_interaction_callback)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/programmer.py", line 106, in start
    self.device_model.start(user_interaction_callback=user_interaction_callback)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/nvmupdi.py", line 70, in start
    self.avr.activate_physical(user_interaction_callback=user_interaction_callback)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.local/lib/python3.13/site-packages/pymcuprog/avr8target.py", line 442, in activate_physical
    return self.protocol.activate_physical(use_reset)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/home/simon/.local/lib/python3.13/site-packages/pyedbglib/protocols/avr8protocol.py", line 292, in activate_physical
    device_id = self.check_response(self.jtagice3_command_response(
        bytearray([self.CMD_AVR8_ACTIVATE_PHYSICAL, self.CMD_VERSION0, int(use_reset)])))
  File "/home/simon/.local/lib/python3.13/site-packages/pyedbglib/protocols/jtagice3protocol.py", line 212, in check_response
    raise Jtagice3ResponseError(error_message, data[0])
pyedbglib.protocols.jtagice3protocol.Jtagice3ResponseError: AVR8_FAILURE_FRAMING_ERROR: Framing error
````

I have it connected as per 10.3.2 in the [user guide](https://www.microchip.com/DS50002787)

What am I doing wrong here?
Do I really need a separate 12V supply, if so how do I hook it up so the mplab snap can trigger a pulse when it needs one

simonschmidt, 2025-02-09:
Nevermind, I think the chip was fried, didn't work it Mplab IPE either. Other chip works fine in IPE, though after IPE upgraded the mplab snap firmware to v2 pmcuprog can't detect it anymore, but that's another problem.

closed 2025-02-09 by simonschmidt

felias-fogg, 2025-02-10:
Hi Simon,

as far as I understood, SNAP does not support high-voltage programming (first sentence in the user manual's Introduction).

Switching to AVR mode (so that pymcuprog becomes usable) can be done with avrdude: https://github.com/avrdudes/avrdude/pull/1596