microchip-pic-avr-tools/pymcuprog #50: debugWIRE gdbserver based on pymcuprog

felias-fogg, 2025-02-08:
Hi,

I am working on a gdbserver for debugWIRE MCUs such as ATmega328P and ATtiny85 using pymcuprog: https://github.com/felias-fogg/dw-gdbserver. Of course, I had to make several modifications and additions to pymcuprog modules, which I all implemented using derived classes in additional modules. This works perfectly fine. However, would you be interested in PRs? 

Best regards,
Bernhard

xedbg, 2025-02-10:
Hi @felias-fogg ,
Its good to see this being explored!
We had an intern dabbling in a GDB server years ago, and that work was published (in not-quite-working form) here: https://github.com/mraardvark/pyavrdebug
The pymcuprog repo here is a "publication" of snapshots of the tool, so PRs are not that simple to be honest.
Is it possible to look at the approach used in the pyavrdebug repo as an alternative to modifying the pymcuprog code itself?  (ie: keep "prog" to "programming"?

felias-fogg, 2025-02-11:
Hi,I took indeed pyavrdebug as a point of departure. It is always good to start with something that is working. Meanwhile, the server code looks quite different, however. What I was referring to, though, were additions or modifications in, e.g., module nvmdebugwire (which is only a stub) and avr8target (class TinyAvrTraget). Currently, I have introduced subclasses of these in order to implement the needed functionality. And I can leave it at that. It would look sort of cleaner to have these parts integrated into pymcuprog, though.Best,BernhardVon meinem iPhone gesendet

closed 2025-02-25 by felias-fogg