microchip-pic-avr-tools/pymcuprog #52: xmega nvm docs missing

mcsmark, 2025-04-11:
Xmega PDI is partly implemented.
The NVM data is missing. And the defined AVR8_MEMTYPEs do not work.

for example :
ERASE_EEPROM = 0x03  # Erase EEPROM only  
does not erase the eeprom. 

I found AVR1612 but that shows different info. i use SNAP programmer.
What i need is either xmega NVM erase implemented or better good documentation of what the protocol expects.
CMD_AVR8_ERASE is used to erase with CMD_VERSION0 and the mode followed by the address of the memory
It works for the whole processor but not for EEPROM. command executes without error but EEPROM is not erased,
Is there some document that describes PDI protocol as used by SNAP ? 
I had a look at avrdude but odd enough that uses different protocol commands.

mcsmark, 2025-04-13:
I found a document for EDBG : https://ww1.microchip.com/downloads/en/DeviceDoc/50002630A.pdf
It does match the python code. But when i try to erase the EPROM using CMD_AVR8_ERASE command with 3 as a parameter. it does not work for Xmega. Works ok for updi chips. But in PDI mode i get no error but the eproms is not cleared. No fuse bit for the eeprom has been set. When i erase the chip it works. Then both flash and eprom are erased. 
And writing eprom pages (erase and write) does not work either.
Since i do not have the source and it does not work i wonder if it is not implemented? Maybe that is the reason the python code also did not implement it?

mraardvark, 2025-04-14:
To clarify: what you are trying to do is erase an XMEGA's EEPROM only using a Snap in "AVR mode"?

mcsmark, 2025-04-14:
yes i am using the SNAP programmer in PDI mode to erase only the Xmega 128A3 eeprom.
After new tests i found that MEMTYPE_EEPROM_ATOMIC will erase and write a page so that kind of solves it. I thought to use it to clear an eeprom by writing pages with FF. But odd enough the protocol does not write a thing when all bytes are FF. There has to be at least 1 byte with a non FF value in order to make the page write. So i can kind of support PDI now but i just do not get why the eeprom can not be erased without getting some kind of error. the actual low level PDI does allow to erase EEPROM. 
I did not tried with other edbg programmers since the goal was to use SNAP programmer to support updi, isp and pdi.

mraardvark, 2025-04-14:
From what I can see it is implemented in the Snap.  However, its plausible that its not tested in that implementation because its not directly accessible from Studio (you can't erase EEPROM only).
It is implemented, and depends upon the values set in the config for XMEGA_EEPROM_BASE and XMEGA_EEPROM_PAGE_SIZE - so check that these are set correctly.

mcsmark, 2025-04-14:
ok thanks. i used the values from the atxmega128a1u.py  other values were hard coded in the py code. 
i will check the atdf file to see if it holds this info and if it is correct.

mraardvark, 2025-04-14:
And note that the address gets added to that which you pass in, so for erasing EEPROM I think you have to pass in 0 as the address in the erase() call

mcsmark, 2025-04-14:
thanks for your help. it turned out i made a mistake with the eeprom base address. i mistakenly used the comment offset for the array instead of the actual offset.
Now it works great :-)