microchip-pic-avr-tools/pymcuprog #55: Register addresses in setting "Device Context" (Table 7-7 in EDBG Guide)

felias-fogg, 2025-08-12:
I am slowly progressing with my GDB server and am now making the first steps into the JTAG territory. In this context, I noticed that the _Embedded Debugger-Based Tools Protocols User's Guide_ does not state which addressing mode to use when specifying the addresses of IO registers in the `device context`  (Tables 7-6 to 7-9). Should it be an IO register address or an SRAM address?

In the `setup_config` method of `MegaAvrJtagTarget` in module `avr8target`, it seems that the necessary registers should all be referenced by their SRAM addresses. This, however, leads to errors. The correct way seems to be:

For `OCDR`, `EEARH`,` EEARL`, `EECR`, `EEDR`, you have to use the IO register addresses, and for `SPMCRS`, the SRAM address. 

It is unclear what the correct address mode of OSCCAL could be. I simply do not have any test cases that I could use to find out whether this address makes any difference. Do you have an answer for that?

xedbg, 2025-08-12:
Hi @felias-fogg,

Indeed you are right with the IO vs SRAM address, and also that it is unclear :|
(SPMCSR is used with an LDI instruction to load via Z while EECR for example is used with IN/OUT.)

The OSCCAL value is something different - I can't remember the details, but I know that it is used in a comparison situation where OCDR and OSCCAL are actually at the same location.  An historic oddity I can't quite remember details of...

So:
- a test case will thus be where OSCCAL == OCDR
- the OSCCAL is surely in the same 'space' as OCDR (IO)

Hope this helps!

felias-fogg, 2025-08-12:
I see! I have noticed cases when two registers shared a memory location in some of the older datasheets. In order to read the alternative register, you had to read twice in succession.